TRI-ML/DDAD #8: Corrupted data

iariav, 2020-10-01:
Hi,
Thank you very much for sharing this dataset.

I tried using this dataset to train your pack-net network, but it seems that some of the datums are corrupted and can't be read from disk correctly (np.load issues an error, and hence training fails).

I have re-downloaded the dataset three times already, and it happened every time.
an example of such corrupted datum is - 
DDAD/000141/point_cloud/LIDAR/1569506040909583.npz

Can you please have a look?
it's currently preventing me from using this dataset for training

Thanks

RaresAmbrus, 2020-10-02:
Hi @iariav 

I just downloaded the data and did some quick checking: both the `train+val` and `test` datasets load properly for me. 

To debug further, can you check that the archives you downloaded are not corrupted by running `md5sum DDAD.tar` (you should get `027686329dda41bd540e71ac5b43ebcb`) and  `md5sum DDAD_test.tar` (you should get `cb244da1865c28898df3de7e904a1200`). 

If that works, can you try running this snippet (for `train+val`) and see if you run into errors:
```
import glob

import numpy as np
from tqdm import tqdm

DDAD_PATH = '/data/datasets/monodepth/DDAD/**/*.npz'

if __name__ == '__main__':
    point_cloud_files = glob.glob(DDAD_PATH, recursive=True)
    assert len(point_cloud_files) == 16600 # train+val
    for f in tqdm(point_cloud_files):
        cloud = np.load(f)['data']
        assert cloud.shape[1] == 4 and cloud.shape[0] > 100 # more than 100 4D points in each cloud
```

iariav, 2020-10-04:
Hi @RaresAmbrus , thanks for the prompt check!
i ran `md5sum DDAD.tar` and got 0ab0d7d5cae67cf2e10e3595471a0c6d so i guess something got corrupted during download. 
i have downloaded the train tar via wget (cmd), and also tried directly clicking the link in your repo and both didn't download properly (wrong checksum). any idea what else i can try? 

p.s
the snippet you provided also fails on the first assert `assert len(point_cloud_files) == 16600 # train+val`. i get that `len(point_cloud_files)` is 18522 and not 16600. perhaps this could hint what is going on?

thanks again

iariav, 2020-10-04:
@RaresAmbrus 
Just a quick update - 
i modified your code a bit by setting `DDAD_PATH = '/data/datasets/monodepth/DDAD/*/point_cloud/LIDAR/*.npz'` and indeed now i get 16600 listings.
the rest of the code still fails though at some point with an error - 

`zipfile.BadZipFile:` Bad CRC-32 for file 'data.npy'

RaresAmbrus, 2020-10-05:
@iariav - thanks, it seems the data gets corrupted while downloading somehow. I could try splitting the DDAD archive into a few smaller files, and maybe that helps. First, let's check if you're able to successfully download DDAD test - that's only around 35GB. Can you please try that?

iariav, 2020-10-06:
Hi,
I downloaded DDAD_test and I don't get the correct checksum. I get 7c93f48e604d945f341fdf907b4eeba1.
I think you are right in assuming that the data gets corrupted during download. 
I will check it with our IT and hopefully we'll be able to solve this.
Thanks for your help.

closed 2020-10-06 by iariav