TRI-ML/DDAD #10: Extrinsic parameters

akhil-robotvision, 2021-03-01:
Dear,

Can you please share the extrinsic camera parameters (mounting position on top of the car), such as height from the ground plane and orientation (yaw, pitch, roll) of the DDAD dataset?

Thank you,

VitorGuizilini-TRI, 2021-06-05:
@RaresAmbrus

RaresAmbrus, 2021-06-08:
Hi, each datum (lidar or cameras) contains an `extrinsics` field that gives its pose with respect to the vehicle frame of reference with origin on the ground below the center of the vehicle rear axle. Is that what you're looking for?

closed 2021-07-21 by RaresAmbrus

AlexD123123, 2021-11-02:
Hi!

I'm looking at the extrinisics data an for example, in scene 000001-camera 01 the translation y has a value of -3143.82.
The y axis according to the drawing should be the "up-down" coordinate, right? (positive is down).

Then in scene 000001 for camera 02 the y translation value is -47463, an order magnitude higher.
In which units is the Y axis given?  

I've checked the translation values across all scenes and x/y/z variables significantly variate.
Do you know why these value change so much ?

Can you please help me better understand these numbers?

Many thanks, 
Alex

ivasiljevic, 2021-11-06:
Hi Alex,

In the raw GT data the poses are in absolute coordinates and thus the values vary significantly between scenes based on where the scene was captured (e.g. on different continents).  The dataloader should convert these into relative camera poses so for any given camera the translations will look rather different from the absolute raw pose.
Can you point to the scene json where you see a large difference within-scene?

AlexD123123, 2021-11-07:
Hi, 

The data makes sense now! Thanks!