TRI-ML/DDAD #15: DDAD trainval is so big, is there a way to download some small videos and calibration params?

jinfagang, 2021-06-01:
DDAD trainval is so big, is there a way to download some small videos and calibration params?

VitorGuizilini-TRI, 2021-06-05:
We have a tiny version of DDAD in the following link (it's used in [packnet-sfm](https://github.com/TRI-ML/packnet-sfm) to run overfit tests). 

s3://tri-ml-public/github/packnet-sfm/datasets/DDAD_tiny.tar

jinfagang, 2021-06-05:
@VitorGuizilini tiny only contains less than 10 images.

RaresAmbrus, 2021-06-08:
@jinfagang I can generate a small version of DDAD train with about 10 scenes -- would that help?

closed 2021-07-21 by RaresAmbrus

jinfagang, 2021-07-26:
@RaresAmbrus  Yes. that definitely helpfull! thank u!

akashchavan15, 2022-09-14:
> @RaresAmbrus Could you please tell me where can I download the DDAD train dataset with 10 scenes which you had generated?

RaresAmbrus, 2022-09-19:
Hi, we have a "tiny" version of DDAD as part of the packnet_sfm repo, does that work: https://github.com/TRI-ML/packnet-sfm#tiny-ddadkitti ?

akashchavan15, 2022-09-19:
@RaresAmbrus tiny version is very small, it has only a few images.

RaresAmbrus, 2022-09-27:
I created a smaller version [here](https://tri-ml-public.s3.amazonaws.com/github/packnet-sfm/datasets/ddad_small.tar.gz) -- this one has 10 scenes and is about 13GB. Can you please check if you can access it? Thanks!

akashchavan15, 2022-09-28:
Yes, I could access it. Thank you so much.

akashchavan15, 2022-10-06:
can anyone please reply to my query? 
Thanks in advance!

RaresAmbrus, 2022-10-06:
Hi @akashchavan15 - sorry I missed your message. Can you please post that on the vidar repo? Thanks!

akashchavan15, 2022-10-06:
@RaresAmbrus thank you for your reply. I have just posted it on Vidar repo. Here is the link 
https://github.com/TRI-ML/vidar/issues/26

Please do reply.