TRI-ML/DDAD #17: LIDAR information

Rolix57, 2021-06-02:
Hi!

Could you share with us what is the LIDAR resolution for asimuth and elevation measurements? as well as its max/min values?

Thx :)

VitorGuizilini-TRI, 2021-06-05:
@RaresAmbrus

Rolix57, 2021-06-07:
I've also noticed that the measurements have been "cleaned" as the number of measurements is inconsistent.
Is there a way of getting the raw measurements?

RaresAmbrus, 2021-06-08:
Hi, the point cloud we store under the `lidar` datum has been accumulated from all the Luminar sensors; we're currently not storing the data / raw measurements from each of the individual sensors. I'm looking into the LIDAR specs (resolution, range) and will get back to you on that shortly.

RaresAmbrus, 2021-06-10:
Following up on this: the resolution of the individual LIDARs is 30deg vertical and 120deg horizontal. The min range is ~2 meters while the max range is up to 250 meters. I hope that helps.

closed 2021-07-21 by RaresAmbrus