TRI-ML/DDAD #19: About the challenge.

jakey-ahn, 2021-06-04:
Hi, I'm participating in dense depth challenge, self-supervised track.
Can I use  ImageNet pre-trained weight for the model for this competition?
I'm aware of the rule that external data should be only monocular videos, but I want to know if the ImageNet weight is available or not.
Thanks!

VitorGuizilini-TRI, 2021-06-04:
Thank you for bringing that up, that's a very good question. 
We have decided to allow ImageNet pre-trained weights, since it is standard in most depth estimation works. That is the only exception though, so other sources of pre-training are still not allowed.

panji530, 2021-06-05:
@VitorGuizilini-TRI  How about using a pretrained semantic segmentation model? This is also commonly used in previously self-supervised methods.

VitorGuizilini-TRI, 2021-06-05:
Although interesting (especially given our [previous work](https://arxiv.org/abs/2002.12319) and inclusion of per-class depth metrics),  unfortunately the use of semantic information has been ruled out for this challenge in particular.

panji530, 2021-06-05:
@VitorGuizilini-TRI  Thanks for the clarification! Can you also answer my question about the extrinsics in another thread?

closed 2021-12-03 by RaresAmbrus