TRI-ML/DDAD #20: Inconsistent extrinsics

panji530, 2021-06-05:
Hi,

Could you advise how to compute the extrinsics between the cameras? I tried this way:
T_cam1tocam5 = np.linalg.inv(data_cam5["extrinsics"]) @ data_cam1["extrinsics"].
Is this the right way?

If this is right, then why the transformation is not consistent across different sequences?

Thanks.

VitorGuizilini-TRI, 2021-06-05:
This should be the right way to do it, yes. What do you mean it's not consistent? There could be some variations in extrinsics between scenes, since the vehicle is recalibrated, is that the case?

panji530, 2021-06-05:
I see. Weird then, because I used such extrinsics to compute the so-called "non-overlapping" masks (with pretrained depth and pose), but those masks are always zero. Any suggestion on how the cross-view photometric loss should be computed?

RaresAmbrus, 2021-06-10:
Maybe @ivasiljevic can shed some light on this as he looked into this most recently.

anh1nt, 2021-07-13:
I tried to find the world coord pointcloud of all views and merge them, but, the view pointclouds are not matched, I think the reason is about extrinsic matrixs

hhhharold, 2022-02-10:
> Hi,
> 
> Could you advise how to compute the extrinsics between the cameras? I tried this way: T_cam1tocam5 = np.linalg.inv(data_cam5["extrinsics"]) @ data_cam1["extrinsics"]. Is this the right way?
> 
> If this is right, then why the transformation is not consistent across different sequences?
> 
> Thanks.

I think the right way should be T_cam1tocam5 =  data_cam1["extrinsics"] @ np.linalg.inv(data_cam5["extrinsics"])

abing222, 2022-02-11:
> > Hi,
> > Could you advise how to compute the extrinsics between the cameras? I tried this way: T_cam1tocam5 = np.linalg.inv(data_cam5["extrinsics"]) @ data_cam1["extrinsics"]. Is this the right way?
> > If this is right, then why the transformation is not consistent across different sequences?
> > Thanks.
> 
> I think the right way should be T_cam1tocam5 = data_cam5["extrinsics"] @ np.linalg.inv(data_cam1["extrinsics"])
extrinsics: cam coor -> vehicle coor, Therefore, the previous formula should be correct

abing222, 2022-02-11:
> I think the right way should be T_cam1tocam5 = data_cam5["extrinsics"] @ np.linalg.inv(data_cam1["extrinsics"])

LionRoarRoar, 2022-02-15:
> I think the right way should be T_cam1tocam5 = data_cam5["extrinsics"] @ np.linalg.inv(data_cam1["extrinsics"])

I tried two ways to build T_cam1tocam5.  T_cam1tocam5 = np.linalg.inv(data_cam5["extrinsics"]) @ data_cam1["extrinsics"] can get right cross-view synthesis result. But in theory,  the right T_cam1tocam5  should be  data_cam5["extrinsics"] @ np.linalg.inv(data_cam1["extrinsics"]). So I got confused. Maybe the provided "extrinsics"  is the inverse of tranformation matrix, i.e. T_wc?

hhhharold, 2022-02-18:
> I tried two ways to build T_cam1tocam5. T_cam1tocam5 = np.linalg.inv(data_cam5["extrinsics"]) @ data_cam1["extrinsics"] can get right cross-view synthesis result. But in theory, the right [...]

I was confused about the extrinsics. In scene 000000, the extrinsics of cam0 that I get from the dataset api (datum['extrinsics']) is [[0.06757921745197824, -0.00886373290436698, 0.9976745379168391, 1.6756778955459595], [-0.9976979761913245, -0.00625200715454027, 0.06752525979420511, 0.28599634766578674], [0.005638942481844948, -0.999941171592379, -0.009265834131195395, 1.5618830919265747], [0.0, 0.0, 0.0, 1.0]]. 
However, I recalculate the extrinsics by qw qx qy qz in raw json file. I got [[-0.009265832960404785, -0.06752522587162985, 0.9976745402236821, 1.6756778955459595], [-0.9999411713437365, 0.006252052685671694, -0.008863728839024532, 0.28599634766578674], [-0.005638988496711739, -0.9976979782019211, -0.067579183929173, 1.5618830919265747], [0.0, 0.0, 0.0, 1.0]].
I am trying to figure out the relationship between two extrinsics.