TRI-ML/DDAD #24: Semantic Labels

csBob123, 2021-07-20:
Hi,
Thank you for providing the dataset. I have a question that do you provide semantic labels for the training and validation images? I saw semantic images on the Github page, but I cannot find any semantic labels in the downloaded data.

RaresAmbrus, 2021-07-21:
Hi - we have panoptic labels (semantic + instance) for the validation set. Were you not able to access them for the validation set? Did you try using the notebook we created ([link](https://github.com/TRI-ML/DDAD/blob/master/notebooks/DDAD.ipynb))?

xiaoning0701, 2021-08-18:
Great thanks for sharing this dataset!! Does the luminar lidar data provide semantic labels?

RaresAmbrus, 2021-11-05:
Hi - sorry but we don't have panoptic labels for the Lidar data, only for the front camera images in the validation set.

closed 2021-12-03 by RaresAmbrus