TRI-ML/DDAD #29: How to get camera relative pose

hcv1027, 2021-11-16:
I'm trying to use camera's ground truth pose to replace PoseNet-sfm's posenet during training.
I use below way to compute camera's odometry for verification, and it looks very strange.
Here is the way I use to compute camera's odometry for a scene (A series of temporal continuous samples).

```python
    scene_idx = 14
    size = 50
    x = []
    y = []
    z = []
    T_cam_odom_gt = None
    for i in range(size - 1):
        datum_0 = dataset.get_datum_data(scene_idx, i, 'camera_01')
        datum_1 = dataset.get_datum_data(scene_idx, i + 1, 'camera_01')
        T_c0_w = datum_0['pose'].inverse()
        T_w_c1 = datum_1['pose']
        T_c0_c1 = T_c0_w.__mul__(T_w_c1)
        if i == 0:
            T_cam_odom_gt = T_c0_c1.copy()
        else:
            T_cam_odom_gt = T_cam_odom_gt.__mul__(T_c0_c1)
        x.append(T_cam_odom_gt.translation[0])
        y.append(T_cam_odom_gt.translation[1])
        z.append(T_cam_odom_gt.translation[2])
            
    plt.scatter(x, z, marker='.')
    plt.savefig('ddad_pose.png')
```

I test it with scene `000014`, since the car turned left at the crossroad, I expect the camera's odometry on x-z plane should something like this one:
[image]

But the output is this one:
[image]

Can you tell me how to get the information I need?

RaresAmbrus, 2021-12-03:
Hi, sorry about these issues. Have you had any luck? For one of our papers we used the `extrinsics` field to compute the relative transformation between sensors, but it's possible that the odometry i.e. the `pose` datum is not accurate for some sequences. I'll try to see what I can find out about that.

GANWANSHUI, 2022-12-05:
Hi @RaresAmbrus , I am also wondering whether the pose by @hcv1027 calculation could be served as the GT pose. Does it reliable? Thanks a lot!