TRI-ML/DDAD #30: Conversion of Data format to Kitti Format

tushardmaske, 2022-01-19:
I have a code running on Kitti dataset where I just pass path to the the main data-file to the code. I want to run the same code on DDAD sataset as it has more clear and wide range of data.

Are/Is there any script/s available to convert DDAD data format to Kitti?  So after running the scripts, I can use DDAD data by just passing the main directory without making big changes in the code.

RaresAmbrus, 2022-03-27:
Hi, sadly we haven't implemented any such conversion scripts, sorry!

Fjolfrin, 2022-06-16:
I do not expect anyone that maintains this repo to implement the contents of this thread, so I just leave some thoughts that would help both the OP, me, and anyone else reading this.
The README mentions that the maximum depth range is 250m. Now, as this is a 360° depth dataset, and the word "range" is specifically used, I suppose it means "radius" and not "diameter". So e.g. the front facing camera depth map will have a maximum depth value of 250m.
KITTI has a maximum depth value of 80m, so one should be fine by capping the DDAD depth maps to 80m. E.g.:

`depth_map[depth_map >= 80.0] = 80.0`

However, the question is how is this 250m maximum depth information encoded to the values of the depth arrays inside DDAD? When I read the depth maps, all values are normalized to [0., 1.], so maximum value is 1.0 and minimum is 0.0. Does this mean that they represent the maximum and minimum depth values respectively? Will a simple normalization to [2, 250] result to correct depth maps?

Petros626, 2026-11-03:
@tushardmaske write it yourself ;)