TRI-ML/DDAD #32: How to download the small version DDAD? @jinfagang I can generate a small version of DDAD train with about 10 scenes -- would that help?

jikerWRN, 2022-03-05:
@jinfagang I can generate a small version of DDAD train with about 10 scenes -- would that help?

_Originally posted by @RaresAmbrus in https://github.com/TRI-ML/DDAD/issues/15#issuecomment-856826124_

Hello! I want to know where can I download the small version DDAD?

jikerWRN, 2022-03-14:
If you want to download the tiny DDAD, please refer to 
# DDAD_tiny
curl -s https://tri-ml-public.s3.amazonaws.com/github/packnet-sfm/datasets/DDAD_tiny.tar | tar -xv -C /data/datasets/

closed 2022-03-14 by jikerWRN

kirilllzaitsev, 2023-10-17:
Trying to load data the following way:
```
ddad_train = SynchronizedSceneDataset(
    DDAD_TRAIN_VAL_JSON_PATH,
    split='train',
    datum_names=DATUMS,
    generate_depth_from_datum='lidar'
)
```
results in:
```
"""
Traceback (most recent call last):
  File "/home/user/.conda/envs/sc/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/user/.conda/envs/sc/lib/python3.9/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/dgp/dgp/datasets/base_dataset.py", line 1277, in _datum_index_for_scene
    return scene.datum_index
  File "/dgp/dgp/datasets/base_dataset.py", line 452, in datum_index
    assert len(datum_key_to_idx_in_scene) == bad_datums + num_datums, "Duplicated datum_key"
AssertionError: Duplicated datum_key
"""
```
due to a mismatch in the number of provided RGBs (35) and LiDAR scans (700)