TRI-ML/DDAD #37: Type of LiDAR data

jbndr, 2023-02-08:
Hello,

I'm not sure in which format the giben LiDAR data is provided.
[image]

Is it just a 64 beam scan or an accumulation?

DongyangHuLi, 2023-04-23:
> Hello,
> 
> I'm not sure in which format the given LiDAR data is provided. [image]
> 
> Is it just a 64 beam scan or an accumulation?

Hi, jbndr
I have the same question. Have you solved it?

jbndr, 2023-04-30:
@DongyangHuLi Hey, the LiDAR pointclouds are dense beams which have been accumulated over multiple sweeps but not accumulated over time like the densified groundtruth in the KITTI dataset.

DongyangHuLi, 2023-05-04:
> @DongyangHuLi Hey, the LiDAR pointclouds are dense beams which have been accumulated over multiple sweeps but not accumulated over time like the densified groundtruth in the KITTI dataset.

Thank you！